darshan-/Battery-Indicator-Support #2: [FR] Power On mark

ildar, 2016-06-03:
I like the app very much and install it on every device I get in my hands :)
I still miss one feature for experimental/unstable ROMs I happen to test. This is the "Power On mark" in LOG, much like we have "Power Plugged/Unplugged" marks.
This is about just one check during the app/service start up.
Thanks for the great app!!!

darshan-, 2016-07-06:
Thanks for the feature request.  I never even thought to log boots, but I can see why it would be useful.  And unless I'm mistaken, it should be trivial to add.  I'll put it on my to-do list for the forthcoming 10.0 release.

darshan-, 2016-07-11:
I just added logging for boot completion to the development version.  Unless something comes up, it should land in a public 10.0.0 release later this week.  Thanks again for the suggestion!

closed 2016-07-11 by darshan-

darshan-, 2016-08-10:
I missed 9.0.0 and 9.0.1; tagging releases and uploading the APK is not a priority to me so I often forget.  But I finally finished up the 10.0.0 release, and thanks to your comment, I remembered to tag it.