darshan-/Battery-Indicator-Support #4: Alarm for when temperature drops below certain threshold

Eloston, 2017-11-24:
Some users use their devices in cold environments, where the battery could get too cold for it to operate well (and cause the device to suddenly power off). If you could add a lower temperature threshold alarm type, that would be great!

darshan-, 2017-11-24:
Thanks for the suggestion!  That's not something I would have ever thought of, as it's not something I've ever run into, but it makes sense.  I've added it to my to-do list.  I'm quite busy with other things at the moment, so I'm not sure when I'll get to it.  But it's an easy thing to add, so it should make it into the next update, whenever I do that.

darshan-, 2018-09-12:
Okay, I just added this to the app.  The update should be out in the next few days for devices running Android 8.0 and up.  I don't think I'll support this feature old older versions of Android unless you, as the filer of the issue, let me know that you're still using older Android, in which cases I'll make that happen.  (For the overall Android ecosystem, 8.0+ is relatively small still, but for this app it's pushing 50% of my users.)

closed 2018-09-12 by darshan-

Eloston, 2018-09-12:
Awesome! Unfortunately, I've since switched to a new device, a new environment, and upgraded to Android 8.1. If I ever run into a similar situation, I'll know what to use ;)

Thanks for following through on the issue!

darshan-, 2018-11-08:
Due to major changes in Android's notification system, it's significantly easier for me to maintain the app with `minSdkVersion 26` for new features at this point.  Backporting this feature to the build for older devices is feasible but not something I have time for right now while I'm extremely busy with other things.  I'll seriously consider adding it when I have a bit of spare time.  Thanks.

robinpaulson, 2018-11-11:
It'd be interesting to set minsdk to 23, but include the new features which require API 26. It could then gracefully fall back if the new feature is not available, while still partially supporting older SDK versions. Is this possible at all in Android?

Josephdecotis, 2018-11-11:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/darshan-/Battery-Indicator-Support/issues/4#issuecomment-437648359",
> "url": "https://github.com/darshan-/Battery-Indicator-Support/issues/4#issuecomment-437648359",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [darshan-/Battery-Indicator-Support] Alarm for when temperature drops below certain threshold (#4)",
> "sections": [
> {
> "text": "",
> "activityTitle": "**robinpaulson**",
> "activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
> "activitySubtitle": "@robinpaulson",
> "facts": [
>
> ]
> }
> ],
> {
> "name": "Add a comment",
> "@type": "ActionCard",
> "inputs": [
> {
> "isMultiLine": true,
> "@type": "TextInput",
> "id": "IssueComment",
> "isRequired": false
> }
> ],
> "actions": [
> {
> "name": "Comment",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"darshan-/Battery-Indicator-Support\",\n\"issueId\": 4,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
> }
> ]
> },
> {
> "targets": [
> {
> "os": "default",
> "uri": "https://github.com/darshan-/Battery-Indicator-Support/issues/4#issuecomment-437648359"
> }
> ],
> "@type": "OpenUri",
> "name": "View on GitHub"
> },
> {
> "name": "Unsubscribe",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 279350714\n}"
> }
> ],
> "themeColor": "26292E"
> }
> ]

darshan-, 2018-11-11:
Yeah, that's how it works in general.  The difference here is that Google has a new policy requiring apps to target 26 now, which has significant changes to how notifications work.  It was much, much less work for me to handle this by having two builds now.  One for 16+ and one for 26+.  Both are maintained and supported with security and bug fixes, but feature development happens for the 26+ version.  It's quite feasible to backport small features like this to the 16+ version when I have some spare time.

robinpaulson, 2018-11-11:
Great, thanks for the info darshan and keep up the good work. Robin

ildar, 2018-11-13:
back to the issue topic: I see one new notification for temp higher then a
threshold, but the OP asked for temp dropping lower.
Version Pro 10.0.4 F-D

darshan-, 2018-11-13:
Version 10.0.4 is the API level 16+ track, which is where we're talking about backporting this feature at some point.  If you're running Android 8.0 or higher, please use version 11.0.2, which is where this feature currently exists.  Thanks.