darshan-/Battery-Indicator-Support #5: Feature request: graphs

RenaKunisaki, 2017-12-30:
Please add a way to view a graph of (charge, temperature, voltage) over time.

ildar, 2017-12-31:
The app can export CSV. Did you think of importing it in any spreadsheet
app and view graphs there?

RenaKunisaki, 2017-12-31:
I was hoping to view them in the app. What app do you recommend for graphing CSV?

ildar, 2017-12-31:
Sorry, I usually analyze such data on my notebook. The one mature
spreadsheet for Android I used is Google's.

robinpaulson, 2018-11-06:
There is a libreoffice viewer for android which may help

robinpaulson, 2018-11-06:
That said, graphs would be a welcome addition, although I appreciate the complexity of adding them.

darshan-, 2018-11-12:
This is something I've periodically fantasized about.  I think it would be fun to do someday, if I have lots of spare time and want a fun side project for a little while.  I'd accept patches if anyone else wanted to get to it before me, but I'd want to be in communication with them to discuss the approach taken.

famo, 2019-12-26:
Unfortunately I can't provide patches, however I want to mention Battery Dog:
http://andbatdog.sourceforge.net/
It does has such a graph feature, which is very compact and comprehensive at the same time:
[image]
The application is also opensource, but over 8 yrs old! So I don't think it is safe to use anymore...

So may this be used as inspiration.