darshan-/Battery-Indicator-Support #7: Android Pie notification priority not working

jcgoble3, 2018-08-29:
I use this app for both low battery alarms and to provide a permanent notification with vital signs on my Google Pixel 2. Because I use Android's native ability to show the battery percentage via the System UI Tuner, I have no use for this app showing the percentage in the status bar, and keep the priority of the persistent notification set to "minimal" to hide the icon. This worked up through Oreo.

However, in Pie, Android no longer supports apps using different notification priorities per notification without the use of Android's notification channels that were introduced in Oreo. A separate notification channel is needed for each priority. Because this app is not currently using channels, hiding the icon can only be done via the system notification settings (the in-app priority selector no longer does anything at all), and doing so forces ALL BatteryBot notifications to minimal priority, which also silences and conceals my low battery alarms.

I need to be able to keep the persistent notification hidden while still allowing alarms to show in the status bar and make noise, and in Pie the only way to do this is to create a separate notification channel for each one. Any chance this could be implemented?

darshan-, 2018-09-12:
Thanks for filing this very clear issue report!  I've got this working well in my development build.  I'll probably push the update in the next few days.

closed 2018-09-12 by darshan-

jcgoble3, 2018-09-12:
Great, I'll be on the lookout for that update. Thanks for your work on one of the most useful apps out there! In particular, the battery logs have been useful to me numerous times in diagnosing problems with various phones.