darshan-/Battery-Indicator-Support #9: Export current values

robinpaulson, 2018-11-15:
At the moment, when I export a log of past events, it includes temperature, voltage and other info, but not current. Would it be possible to include this?

darshan-, 2018-11-16:
My thinking was that the current changes so rapidly that it wouldn't be too useful to log.  It would be straightforward to start logging it if folks think it would be useful to them.

robinpaulson, 2018-11-16:
It's generally fairly stable on my device, at first after I plug in it tries a few different current rates before settling on one, then stays at that value until it is close to full - but not always. I'm trying to find out if there's a correlation between current, temperature and how charged the battery is, so knowing the current over a longish period is very useful to me.

darshan-, 2018-11-16:
Okay, that's good enough to convince me.  I'll add it when I next sit down to add features, which will probably be January, I'm guessing.

robinpaulson, 2018-11-25:
Marvellous, cheers darshan