darshan-/Battery-Indicator-Support #10: Show current drain

robinpaulson, 2018-12-08:
I'm not sure if this is possible, but it'd be nice to have. Is there anyway to read how much current the device is using?

Cheers

darshan-, 2018-12-13:
Hi,

On devices that support it, the Pro version includes a "current hack" (I call it a hack because it's not officially supported by Android, it bypasses Android to get the information, and it works on many devices, but far from all devices) that shows this information.

closed 2018-12-13 by darshan-

robinpaulson, 2018-12-14:
Ah, I think we're talking about different things. I have the current hack enabled already, but that only shows the current flow into the battery from an external power source. At the same time as the battery is charging, the phone is also discharging it. I'd like to know that rate, the current going from battery to phone. I might draw a circuit diagram if necessary.

darshan-, 2018-12-14:
Ah, I think I understand.

My understanding of the value used is that it represents the net current to/from the battery.  I don't believe Linux exposes (nor probably can detect on typical devices) current specifically from battery to phone while charging.  (While off the charger, of course what you see is from battery to phone.)  If you find any way to do this, I'm certainly into the idea of exposing it in the app!  It'd be cool to show charger-to-battery and battery-to-device current separately.

robinpaulson, 2018-12-18:
I didn't realise it was net drain, it always seems so stable after the first few seconds. I might look up kernel docs to find out. Any suggestions where to start? Kernel.org perhaps?

darshan-, 2018-12-18:
It's certainly possible it works differently on different devices.  But I believe the intention and the normal use case is net current.

This might be a starting off point:

https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-power

And then look at the kernel source for your device to be sure what's going on when the app reads `/sys/class/power_supply/<supply_name>/current_now`.