darshan-/Battery-Indicator-Support #12: [FR] Minimal Mode (no Status Bar icon, no Notification)

famo, 2019-12-26:
Hi, I really like this app.
However I mainly use it for logging purposes or to look something up when there are issues. So I don't need continuous information from the app. Therefor it would be nice, if one could turn off the status bar icon and the notifications.

I have seen you already have lots of option to configure these, it is just missing an off option! :smiley: 
Thanks!

fohfuu, 2022-01-04:
Seconding. I only need it to see how much power it's drawing right this second, not over time. It would be perfect if I could stop it asking me to turn off battery optimisation any time it opens.

onchov, 2025-04-17:
I have no problem with notifications, as I can manage them at the system level, but having the option Settings > Status Bar Icon > Icon Set > **Off** (in addition to Classic, Plain Number and Smaller Number) would be great. Thank you.