darshan-/Battery-Indicator-Support #16: Widget broken on Samsung J5

adeak, 2020-12-21:
I love BatteryBot, it's been the single most reliable app I've used on all my smartphones :) Thanks for the great app!

I'm using the free app. I updated it a few days ago, to version 11.0.4 (December 4 update according to google). This broke the widget altogether: the widget icon just shows a black box with the text "Problem loading widget". This happens both with a pre-existing widget and when trying to create a new widget.

I'm on a Samsung J5 2017 (SM-J530F) running Android 9 and One UI version 1.1.

Screenshot of the broken widget:
[image]

darshan-, 2020-12-21:
Hi, thanks for the bug report!

A couple basic things that might, if we're lucky, resolve the issue:

1) Remove all BatteryBot widgets from your desktop, then
2) Open up the system settings, go to Apps, find BatteryBot, and clear all storage (not just cache) for the app.
3) Try adding another widget.

If it works, great!  If not, you can try uninstalling BatteryBot and then reinstalling it.

If that still doesn't work, and you're able and willing to install development builds of the app, please email me, we can try to narrow down where things are going wrong and then fix it.  (The widgets are working for me on my test devices, so it's not obvious to me how to fix this without being able to try things on the device known to have the issue.)

Thanks!

adeak, 2020-12-21:
Thanks, @darshan-. Clearing the widget and cache and data didn't help. Reinstalling the app, however, did! I should have thought of that myself :) Thank you for the quick response.

closed 2020-12-21 by adeak