darshan-/Battery-Indicator-Support #17: Repeat alarm

hboetes, 2021-07-01:
I might be missing something, but I can't find how to let alarms repeat. Under some circumstances you simply miss the alarm and that can easily be solved by adding a repetition option.

I hope you like the idea.

saxofooon, 2021-12-18:
I had the same idea. When the phone is charging it is likely that you are not nearby all the time. So about 10 repeated alarms, every 2 minutes would be great. Unfortunately the charging cannot be limited without root.

saxofooon, 2022-01-06:
Workaround: can you please set the alarm when charging exceeds x% not in 5% steps but in 1% steps? So you can add multiple alarms when charging exceeds e.g. 79%, 80%, 81%, 82%...